am32-firmware/AM32 #9: The temperature algorithm with ARTERY mcu maybe wrong

ymzcdg, 2024-02-13:
In "main.c", when my mcu is from ARTERY, the function to calc temp is "getConvertedDegrees() in ADC.c".In datasheet of AT32F421 ,the algorithm is "temp(°C) = {(V25 – VTS) / Avg_Slope} + 25"and Typ_Avg_Slope = -4.3mV, V25=1.28mV.
So if my mcu is AT32F421K8U7,I will use "(12800 - (int32_t)adcrawtemp * 33000 / 4096) / -43 + 25;" to calc temperature inside mcu.
Please answer my doubts, thanks.

DJ-URAN, 2024-02-16:
4096? Are you sure? Maybe, 4095?

ymzcdg, 2024-02-17:
> 4096? Are you sure? Maybe, 4095?

In ”Mcu/f421/Src/ADC.c“ it's 4096. Maybe 4096 is the correct value, so I did not change it. I think 4095 is correct value in math ,but in computer, calculate /4096  is  faster then /4095, and the error is very small.

AlkaMotors, 2024-02-17:
It doesn't really matter anyway since the temperature sensors are not calibrated like stm32.  Many vendors use a external ntc for artery mcus.

ymzcdg, 2024-02-18:
> It doesn't really matter anyway since the temperature sensors are not calibrated like stm32. Many vendors use a external ntc for artery mcus.

I want design a hardware, what type of sensor does your formula correspond to?My NTC sensor calculation formulas all have logarithmic functions, but your formula does not have logarithmic functions.I think the sensor corresponding to your formula is easy to use, and I would like to know its model.
Thanks.

lida2003, 2024-08-08:
> Many vendors use a external ntc for artery mcus.

Are there any recommended constraints or suggested positions for placing the NTC (Negative Temperature Coefficient) sensor?