am32-firmware/AM32 #10: Release assets .hex flashing question

bisak, 2024-03-07:
Hey, apologies for asking here, couldn't join the Slack / Discord, which I realise would be a far better place.

Question: Do the .hex files on the [release page assets](https://github.com/am32-firmware/AM32/releases) contain the bootloader and EEPROM, should they be flashed at 0x08000000 via ST-Link as one? Or the bootloader [from here](https://github.com/AlkaMotors/AM32_Bootloader_F051/tree/main/Bootloaders) / EEPROM should be flashed at 0x08001000.. and 0x08007C00 separately?

Thanks

AlkaMotors, 2024-03-07:
All, you need to do is flash with st-link is the bootloader that matches the input pin and mcu if its a blank esc. 

Once the bootloader is flashed, use the config tool linked in the readme to flash the firmware that matches your esc. Make sure you are on the "flash" tab. connect to the esc and flash.

bisak, 2024-03-07:
Thanks for the speedy reply! Can I use Betaflight passthrough and esc-configurator instead of the config tool, after flashing the bootloader?

kedeng, 2024-09-06:
> Thanks for the speedy reply! Can I use Betaflight passthrough and esc-configurator instead of the config tool, after flashing the bootloader?
I think you can use [AM32 Configurator](https://am32.ca/), see more info https://oscarliang.com/flash-update-am32-esc/