am32-firmware/AM32 #11: VTOL motor positioning on stop

okiaba, 2024-04-08:
Hi there, awesome work!
This is a feature request or more even a feature possibility request.

I'm looking for a way to align VTOL Quadplane Motors in flight when they are stopped. (like this: https://youtu.be/rsTkrPtcVTk?t=34) From my understanding this should be possible with a simple hall sensor+magnet (on/off input) and some software tweaks to have a slow turning movement until the hall sensor is reached?

There are very few ESCs that have this capability and none of them fit my criterias, bringing this feature to AM32 would be a huge win for the VTOL community, as there so much more Hardware available.

What do you think? Would such a feature be at all be possible and/or of your interest?

Thanks!

AlkaMotors, 2024-04-14:
Hi, okiaba, sorry for he delayed response. Hall sensors are not suitable for aligning a motor on high pole count motors. If the motor is 12 or 14 pole then there are 6 or 7 electrical revolutions per physical revolution so there would be 6 or 7 positions that would have the same hall output. You would need some type of encoder with 0-360 degree steps to find the actual rotor position.

okiaba, 2024-04-15:
Thanks for the reply! Sorry that I was clear in my explanation. I'm not about the traditional hall sensors on a sensored brushless motor, rather than a special motormount with and one hall sensor and one magnet that aligns with the propeller, so that there is only a short "on" phase of the hall sensor when the propeller is aligned (two magnets are obvs. also possible for both propeller blades) - so we would need one additional digital input, for which the telem line could even be used, as ardupilot (the FC FW) supports bdshot telem. Does that make sense?

okiaba, 2024-04-15:
The stopping mode could maybe work as follows:
- Brake the motor.
- Start it again with slow, blind sinus commutation until hall sensor reads 1 and 0 again.
- drive backwards with very slow blind sinus commutation until hall sensor reads 1 again
- brake motor (and optionally monitor hall sensor continuously, if it goes to 0 again, repeat the process)

AlkaMotors, 2024-04-15:
Yeah, ok this is very simple to do. We can easily align to a pulse out from a hall sensor reading a single magnet.

okiaba, 2024-04-15:
Okay awesome! Let me know how I can support the development! I can definetly build a test setup to test the code.

criegit, 2024-11-07:
this would actually be awesome. 

Addition: Possibility to use a rotational magnetic sensor with absolute angle output via i2c/comparable.
But the hall sensor would already be great in the first place.