am32-firmware/AM32 #16: A flash overflow error occurred while compiling the E230 firmware

Ahdjdus, 2024-04-28:
I tried to compile the firmware with the included arm-none-eabi-gcc is '13.2.1', but when I compiled the e230 there was a space overflow, everything else worked fine at compile time, I suspect it is a problem with the GD32E230K8_FLASH.ld file, but I don't know where to start. Here's what you need to know when you compile the e230

```
hao@Windows-10 MINGW64 /c/project/ESC/Firmware/AM32-2.08
$ make e230
Compiling AM32_GD32DEV_A_E230_2.8.elf
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/thumb/v6-m/nofp\libc_nano.a(libc_a-closer.o): in function `_close_r':
closer.c:(.text._close_r+0xc): warning: _close is not implemented and will always fail
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/thumb/v6-m/nofp\libc_nano.a(libc_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text._lseek_r+0x10): warning: _lseek is not implemented and will always fail
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/thumb/v6-m/nofp\libc_nano.a(libc_a-readr.o): in function `_read_r':
readr.c:(.text._read_r+0x10): warning: _read is not implemented and will always fail
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/thumb/v6-m/nofp\libc_nano.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x10): warning: _write is not implemented and will always fail
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/thumb/v6-m/nofp\libc_nano.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x10): warning: _write is not implemented and will always fail
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: obj/AM32_GD32DEV_A_E230_2.8.elf section `.text' will not fit in region `FLASH'
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: section 
.firmware_version LMA [08007bd0,08007bdd] overlaps section .text LMA [080010cc,080093df]
C:/Application/pack/arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 7540 bytes
Memory region         Used Size  Region Size  %age Used
           FLASH:       35188 B        27 KB    127.27%
   FLASH_VERSION:          14 B         14 B    100.00%
       FILE_NAME:          30 B         32 B     93.75%
          EEPROM:          0 GB         1 KB      0.00%
             RAM:        3728 B        16 KB     22.75%
          EEPROM:          0 GB         1 KB      0.00%
             RAM:        3728 B        16 KB     22.75%
collect2.exe: error: ld returned 1 exit status
make[1]: *** [makefile:103: obj/AM32_GD32DEV_A_E230_2.8.elf] Error 1
make: *** [makefile:88: GD32DEV_A_E230] Error 2
```

Bi-an-NEI, 2024-08-02:
it may be need fix flash size at mcu/e230/GD32E230K8_FLASH.ld. flash size  of e230 is 64kb in fact, but that only 27kb in Flash.ld

tridge, 2024-08-23:
this is fixed in main now