am32-firmware/AM32 #19: EDT support?

fschuindt, 2024-06-02:
Is EDT on DShot600 supported? Moreover, is there an official documentation page? Or list of features?

Thank you!

fschuindt, 2024-06-03:
It seems it does, as [found on the Discord server](https://discord.com/channels/1029170737342316605/1029170737342316608/1215411619950239855).

closed 2024-06-03 by fschuindt

howels, 2024-06-03:
> It seems it does, as [found on the Discord server](https://discord.com/channels/1029170737342316605/1029170737342316608/1215411619950239855).

EDT and dsho600 is supported, please refer to the betaflight docs
https://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware#am32

shrinibas7, 2025-04-28:
Does the motor phase back-EMF signal go through a comparator circuit before reaching the MCU, or is it connected directly with just a resistor