am32-firmware/AM32 #20: STM32G031 support

zyura, 2024-06-12:
I successfully built and ran AM32 on my custom board using the STM32F051 microcontroller, and it's working fine (thanks for your work!).

Would it be feasible to run AM32 on the STM32G031G8U6? These microcontrollers come in a smaller 4x4 package and are more cost-effective compared to the F051.

While there's a target for the F031, there isn't one for the G031. If you believe it's possible to adapt AM32 for the G031, I'm willing to modify my board accordingly, conduct testing, and submit the necessary patch.

Regards.

zyura, 2024-06-12:
There's "[Migration of applications from STM32F0 Series to STM32G0 Series](https://www.st.com/resource/en/application_note/an5145-migration-of-applications-from-stm32f0-series-to-stm32g0-series--stmicroelectronics.pdf)" app note discussing the subject.

So basically my question is about possible nuances that would make it hard or impossible to run AM32 on G031.

zyura, 2024-06-12:
When using its options to use PA9 instead of PA11 and PA10 instead of PA12 the pinout becomes compatible with HARDWARE_GROUP_F0_A.

[image]

AlkaMotors, 2024-06-12:
G031 has no comparators. G071is compatible and supported.  The G051 will work too.

zyura, 2024-06-12:
You're right, didn't notice this. Thanks.

closed 2024-06-12 by zyura