am32-firmware/AM32 #23: Add S.Port / FBus

jasc76, 2024-06-20:
Hi there
can you add the possibility to output S.Port and FBus telemetry? It is a really nice thing to buy opensource esc's and have telemetry directly from the esc and AM32 sadly doesn't have this

jasc76, 2024-09-23:
Any word on this feature?

freasy, 2024-09-23:
This is not on our priority list. Keep in mind that we have limited flash.

M-i-c-h-a-e-l1, 2024-10-01:
Maybe the following will help: [ESCape32](https://github.com/neoxic/ESCape32):

- Servo PWM, Oneshot125, automatic throttle calibration
- DSHOT 300/600/1200, bidirectional DSHOT, extended telemetry
- Analog/serial/iBUS/SBUS/SBUS2/CRSF input mode
**- KISS/iBUS/S.Port/CRSF telemetry**
- DSHOT 3D mode, turtle mode, beacon, LED, programming
- Sine startup mode (crawler mode), brushed mode
- Proportional brake, drag brake
- Temperature/voltage/current protection
- Variable PWM frequency, active freewheeling
- Customizable startup music

AlkaMotors, 2024-10-01:
We have chosen to keep the number of auto detected input types at a minimum to reduce the chance of unwanted spoolup and misdirected signals. We will probably stick with Pwm, dshot and CAN, and crsf to keep it as safe as possible.

jasc76, 2024-10-02:
Ok if you don’t add FBus, but S.Port telemetry would be great given the large userbase

M-i-c-h-a-e-l1, 2024-10-02:
As my post to your reply on GitHub was deleted, maybe email works ...
Please have a look at ESCape32 on GitHub. This should have all the features
you are looking for.

Kind regards
M

freasy, 2024-10-02:
> Ok if you don’t add FBus, but S.Port telemetry would be great given the large userbase

We discussed it and agreed on adding s.port telemetry. Are you in the discord? I would appreciate it you could test it

freasy, 2024-10-03:
@jasc76 on what ESC are you wanting this? Then i can send you a hex file to test.

Satcomix, 2024-11-07:
Hello freasy,
I would like to help test the FrSky S.Port protocol.
I own several FrSky transmitters X20S, X20Pro, TandemXE and receivers up to 18 channels with S.Port or FBUS
I get several AM32 ESCs with voltage, current, capacity, temp, rpm sensors.
Sequre SQESC2670 70A G071
SQESC28120 120A G071
NeutronRC 70A G071
NeutronRC Exuav 70A F051
VIM Drones L431 AM32 development board with A1404 BL-Motor
VIM Drones L431 Drone CAN ESC Nano 15A.
I can also be found on Discord.
Can you send me please some .hex files for testing the S.Port protocol on my boards.
Best regards,
Torsten

robthomson, 2024-11-07:
Happy to test too..

jasc76, 2024-11-07:
> @jasc76 on what ESC are you wanting this? Then i can send you a hex file to test.

Oh sorry, late to the party 🙃
Sequre 28120

Satcomix, 2024-11-07:
> > @jasc76 on what ESC are you wanting this? Then i can send you a hex file to test.
> 
> Oh sorry, late to the party 🙃 Sequre 28120
> 
> Sure, I can join your discord

Hello jasc76,
it`s never to late for S.Port telemetry :-))
https://discord.com/channels/1029170737342316605/1304092323646406719
see you,
greatings,
Torsten

jasc76, 2024-11-07:
TBS discord !?

I won’t be able to best before the weekend. Will try it ✊🏻

jasc76, 2024-11-11:
> Hello jasc76, it`s never to late for S.Port telemetry :-)) https://discord.com/channels/1029170737342316605/1304092323646406719 see you, greatings, Torsten

Hmm
Can’t do anything with this discord link 🤷🏼‍♂️ its just the TBS general chat

freasy, 2024-11-11:
> TBS discord !?
> 
> I won’t be able to best before the weekend. Will try it ✊🏻

no, am32 discord

Satcomix, 2024-11-13:
> Hmm Can’t do anything with this discord link 🤷🏼‍♂️ its just the TBS general chat

Hello jasc76,
Here is the discord link from the GitHub AM32 mainpage.
https://discord.gg/h7ddYMmEVV
Under #development there is the "S.Port" channel.
hope to see you there.

jasc76, 2026-03-22:
For me the SPort from here works
https://github.com/pdingemans/AM32_SPORT/releases/tag/3.0.x

It would be really great if you could merge it !!!!!

Satcomix, 2026-03-22:
> For me the SPort from here works https://github.com/pdingemans/AM32_SPORT/releases/tag/3.0.x
> 
> It would be really great if you could merge it !!!!!

Hello Jasc76, 
thank you very much for the tip about this very good news regarding the S.PORT protocol on the G071 chipset for AM32. 
I will test it extensively over the next few weeks. 
Best regards, Torsten

M-i-c-h-a-e-l1, 2026-03-22:
On the Sequre 2670 make sure that a 100nF capacitor is on the board. Otherwise the current measurement will be completely off.

Kind regards
M.

jasc76, 2026-03-23:
> On the Sequre 2670 make sure that a 100nF capacitor is on the board. Otherwise the current measurement will be completely off.
> 
> Kind regards M.

Do you have a pic where that cap is supposed to be !?

jasc76, 2026-05-11:
Any news on this?
Will it be incorporated in the standard releases???

jasc76, 2026-09-09:
Hi
Any movement on this topic?