am32-firmware/AM32 #24: AM32 FOR GEPRC STABLE PRO F7 ESC BL32 2-6S 4in1

miguelportilloyaluk, 2024-06-25:
Hello, I have a GEPRC STABLE PRO F7 stack with the 4 in 1 esc that runs with BLHELI32 firmware (Target: GEPRC-BL32-4IN1). I want to upgrade my esc to AM32 firmware, but I could not find the AM32 target for it. Could someone tell me if there is an AM32 target for the STABLE PRO F7 ESC BL32 2-6S 4in1? Is there anyway you can help me find it?. Thanks. Miguel

GrayMagellan, 2025-01-18:
> Hello, I have a GEPRC STABLE PRO F7 stack with the 4 in 1 esc that runs with BLHELI32 firmware (Target: GEPRC-BL32-4IN1). I want to upgrade my esc to AM32 firmware, but I could not find the AM32 [...]

Hi Miguel! I have same ESC and I search AM32 firmware for this ESC. If anybody can help me - please

miguelportilloyaluk, 2025-01-20:
Hello, how are you? I wrote to GEPRC and they told me this:

 

 

Just use the stack as it is, forget about AM32 with this stack, it works great anyway.  

 

Ing. Miguel M. Portillo

Enviado el: sábado, 18 de enero de 2025 08:37 p. m.
Para: am32-firmware/AM32 ***@***.***>
CC: miguelportilloyaluk ***@***.***>; Author ***@***.***>

AlkaMotors, 2025-01-20:
https://github.com/am32-firmware/AM32/releases/download/v2.17rc1/AM32_GEPRC_4IN1_F421_2.17.hex

miguelportilloyaluk, 2025-01-21:
Thanks a lot

 

   Ing. Miguel M. Portillo

          Oficina Tecnica

      Cel.: 0984-453-100

Enviado el: lunes, 20 de enero de 2025 05:45 p. m.

 

https://github.com/am32-firmware/AM32/releases/download/v2.17rc1/AM32_GEPRC_4IN1_F421_2.17.hex