am32-firmware/AM32 #25: FW Info missmatch AM32_WRAITH32_F421_2.12 reports as NEOESC f051 FW Rev: 1.35

menschel, 2024-06-27:
I recently flashed AM32 onto Holybro Tekko32 F4 Single ESC according to [guide](https://github.com/am32-firmware/am32-wiki/blob/main/docs/guides/Flashing-Tekko32-45A-Single-ESC.md).

I used "AM32_WRAITH32_F421_2.12.hex" firmware and as far as I can judge the result, the esc correctly works on a motor.

The strange thing is that the "Esc_Config_Tool_1_82" tool lists firmware name and revision with

"NEOESC f051 FW Rev: 1.35"

AlkaMotors, 2024-06-27:
Try connecting again and see if its correct now. The firmware name is not set until the first time the firmware boots. When you hear the beeps it will set the firmware name then.

menschel, 2024-06-27:
Nope, does not update the FW Info. Bench setup is

4S Lipo + ESC + 2600kV Copter Motor + Arduino

I found out that I needed 4S to make it actually beep, the small 3S I was using before did not make it beep.

I assume the Artery bootloader does something else then the STM bootloader.
I did build Keil Project for 64k bootloader and flashed that one because I have the 64k chip.

Console output also looks weird with the F3ESC, should be F4.

```
$ ./SerialPortConnector.sh
number of ports : 2 
number of ports : 2 
number of ports : 2 
number of ports : 2 
size of data : 9 
F3ESC_2KB_PAGE
good ack !!!!
size of flash : 51 
good ack read !!!!
GOOD crc FROM ESC -- read
 output integer 0
good ack !!!!
size of music : 131 
good ack read !!!!
```

In general everything works, so please consider this a nice to have fix. Thanks.

AlkaMotors, 2024-06-27:
OK that's the problem.. the firmware is saving the name to the 32kb location. You must flash the 32kb bootloader.

AlkaMotors, 2024-06-27:
[image]

menschel, 2024-06-27:
That solved the issue. It now says Wraith32_F4.

Thank you very much for your help. 

Please let me know if I can support the project in any way.
I plan to make an update to the tutorial together and will likely make a PR later. Now that I know my mistakes, it should be fairly simple.

closed 2024-06-27 by menschel