am32-firmware/AM32 #32: Hardware manufacturer's firmware release proposal

BAYCKRC, 2024-07-09:
Hello! Numerous developers of AM32

Now that BLHeli32 has ceased operation and AM32 is rapidly developing, many hardware manufacturers will flock in. At this time, more and more ESC models will need to be released. With the current AM32 version release mechanism, hardware application information will be submitted and approved, but firmware will need to be synchronized and released after the next new version is released. Therefore, it is recommended to use a management method similar to betaflight and expresslrs.

Example：https://github.com/betaflight/betaflight   and   https://github.com/betaflight/config
Example：https://github.com/ExpressLRS/ExpressLRS   and  https://github.com/ExpressLRS/targets

freasy, 2024-07-09:
Hello,

thank you for your concern. We are aware of this, and we have plans to simplify this.
In the near future, some parts of the code will be refactored to make things easier for us to maintain and also add new targets.

Kind regards
un!t

BAYCKRC, 2024-07-09:
I hope to arrive early this day, thank you all for your hard work!

BAYCKRC, 2024-07-09:
@freasy May I ask, what is the application process for the new ESC firmware? I need the help of your developers now, and we can easily contact each other on Discord