am32-firmware/AM32 #39: Firmware Selection for VGoodRC 30A lite ESC (F051K66 with MP6531AGR)

menschel, 2024-07-18:
Hello,

I'm reworking a Diatone Tyrant S 2017 build and want to flash AM32 onto the 30A version of these ESCs.

[Thread on RCGroups](https://www.rcgroups.com/forums/showthread.php?2606140-Free-Testing-VGOODRC-ARM-32bit-ESC%21)
Pictures
[front](https://static.rcgroups.net/forums/attachments/6/4/0/2/3/8/a8714250-201-DH%20ESC%20PCB%20SIZE%201.jpg)
[back](https://static.rcgroups.net/forums/attachments/6/4/0/2/3/8/a8714251-96-DH%20ESC%20PCB%20SIZE%202.jpg)

I reverse engineered that it's standard STM F051 with MP6531AGR, signal input is via PA2 via 480Ohms resistor.

So I would take the [STM32_F051_BOOTLOADER_PA2_V11.bin](https://am32.ca/api/file/Ym9vdGxvYWRlcjpTVE0zMl9GMDUxX0JPT1RMT0FERVJfUEEyX1YxMS5iaW4=) and then try to flash
[AM32_MP6531_F051_2.12.hex](https://am32.ca/api/file/cmVsZWFzZXM6djIuMTIvQU0zMl9NUDY1MzFfRjA1MV8yLjEyLmhleA==).

Anyone already flashed these?

menschel, 2024-08-17:
I'd like to share the conclusion on this matter. The F051K6 on that ESC is apparently RDP2 protected. STM32CubeProgrammer does not connect to the target. Nothing via SWD, NRST and Boot0 have no effect.

closed 2024-08-17 by menschel

AlkaMotors, 2024-08-17:
Rdp2! That's not good. Most esc just have level 1 and can still have the option bytes reset.