am32-firmware/AM32 #40: skystars am60 (MVP60) wrong current in telemetry

Stricted, 2024-07-19:
i recently got a skystars am60 (MVP60) ESC

after i hooked up the telemetry line i can see in mission planner that the esc is reporting the wrong current for all 4 esc
[image]

the voltage is within margin of error, the current value is completely out of whack, when connected to an external power supply and measuring current with a good multimeter i can dial it in to draw a total of 2A (both multimeter and power supply agree) but the ESC reports >60A of current draw on all 4 esc

any pointers as to what is going wrong here or how i can debug it? i already updated the esc to the latest firmware (2.12)

AlkaMotors, 2024-07-19:
There is no current sensor feedback on those esc's. There are only a handful of older esc's I know of with individual current shunt. Most 4 in 1s only have the current sense going to thr fligth controller.

Stricted, 2024-07-19:
thanks,

interesting, so the current sense line is not connected to the individual esc to read? damn

would it be possible to have it just report 0 in such cases? it reporting 240A of total current in the mission planner manitor is not great
[image]

guess i have to read into how to use the current sense wire

howels, 2024-07-20:
> thanks,
> 
> interesting, so the current sense line is not connected to the individual esc to read? damn
> 
> would it be possible to have it just report 0 in such cases? it reporting 240A of total current in the mission planner manitor is not great ![Screenshot_38](https://private-user-images.githubusercontent.com/4091731/350506841-df3e0924-55d6-44e3-8eaa-c7d5118df941.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1MTMzODUsIm5iZiI6MTcyMTUxMzA4NSwicGF0aCI6Ii80MDkxNzMxLzM1MDUwNjg0MS1kZjNlMDkyNC01NWQ2LTQ0ZTMtOGVhYS1jN2Q1MTE4ZGY5NDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjBUMjIwNDQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTNlNjhhZTlkNGZjODg2YmYzMDY1ZGVhN2E2ZTNmYmViNWIwY2I3YTExZjVkMGI5NWNlNDUyODI2OTIxNjA3YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.keNZfQ_b7cBwJzeUsQhPKnl_kFdG6T3OEofrCm5zzNY)
> 
> guess i have to read into how to use the current sense wire
> 
> edit: i also see no configuration/compile target for neither `AM32_SKYSTARS_AM60_V2_F421` nor `AM32_SKYSTARS_MVP60_F421`, am i missing something?

Usually the current sense wire on the ESC has a dedicated ADC pad on the FC which scales the 0-3.3V value and calculates current usage. If a dedicated Current ADC is absent it can sometimes be possible to use the RSSI ADC pad instead.