am32-firmware/AM32 #43: Target groups - pin maps for custom boards?

piotr-defendeye, 2024-07-22:
Hey! I'd like to design a custom ESC, I've successfully made several BLHeli-S based ESCs before, and would love to take a shot at making an AM32 one! Is there any documentation or reference schematics for MCUs other than F051 describing what are the pin mappings for specific target groups? Which pins are feedback, which are for the gate driver, etc? I'm specifically looking for any references for the GD32E230. Thanks!

AlkaMotors, 2024-07-23:
[standard_esc_1_1.pdf](https://github.com/user-attachments/files/16355223/standard_esc_1_1.pdf)
Here is a schematic that works for a few esc's.  The gigadevice e230 is the dame pinout as the f051 or At32f421 .

mayl, 2024-07-28:
Thanks for sharing that schematic.  Is there a way to wire it for pin compatibility with the STM32G071? I was working on a schematic and noticed that part seems to be "off by one" compared to the F051...

[image]
[image]

mayl, 2024-07-28:
Well, I'm guessing the answer is probably not...
[image]