am32-firmware/AM32 #52: iFlight G071 E55 4in1 64k bootloader

xagos, 2024-08-01:
I have unlocked and erased an iFlight Blitz E55 esc with STlink in Stm33Cube, to flash it with the last G071_64K Bootloader.
After that process, the esc(s) are not recognized by all Firmware update tools (Am32.ca etc.) I have tested.
Changing the bootloader to the "not 64KB" (v8) version seems to work - firmware can be flashed via FC passthrough, but some major information/feedbacks (description in the green boxes on the AM32.ca page) are missing.

xagos, 2024-08-02:
Today I have testes the behavior of the actual Firmware "AM32_IFLIGHT_BLITZ_G071_2.12" on my iFlight Blitz E55 G071:
the motor are emitting a start sound (beep), but I am not able to connect betaflight to the ESCs - pass through (flashing firmware etc) is working, but I do not have the expected information in the AM32.ca config tool:
[image]

When I use  the ESC_Config_tool I get:
[image]
but I am not able to drive the motors...

Is there a way to debug this?
Do I have to install Keil and compile a specific version for my ESC?
I thought, that the iFlight BLITZ E55 would run with the precompiled firmware out of the box, after installing the bootloader.

Regards...

AlkaMotors, 2024-08-02:
There are two g071 versions of the iflight blitz. There are ones made with the g071 and ones made with the g071N variant. If you have the g071 standard version It uses the pb4 bootloader and g071 blitz firmware. Currently you have a pa6 input pin bootloader flashed but the firmware for the pb4 input.

AlkaMotors, 2024-08-02:
The blitz firmware for the n variant is not released yet. Can you join the discord and we can give you test version for that esc.

xagos, 2024-08-02:
Ok - the PB4 bootloader was my fist try, but than no esc was found. So PA6 is installed but the firmware is the wrong version. Never ever have used Discord before, but I have created an account now... Tanks for your support ;)

How do I get the .hex-File...?

bhuism, 2024-08-05:
@xagos I think you can download it here: https://am32.ca/downloads

xagos, 2024-08-06:
Thank you - I had access to the file via discord and I have successfully tested the G071N version (2.16RC3) for the iFlight K55 ESCs. Regards...

xagos, 2024-08-06:
iFlight E55 G07N Test

Bootloader: STM32_G071_BOOTLOADER_PA6_V8.hex (PA6 for the "N" Version, there are variants of the iFlight K55 with a different pinout)
Firmware: AM32_IFLIGHT_BLITZ_N_G071_2.16.hex (RC3)

- erased Flash and programmed bootloader with the STLink and STM32Cube (linux) (but used STLink (win) to unlock the flash)
- flashed firmware with the ESC_Config_Tool_1.82 (AM32.CA did not run reliably)
- loaded default settings
- changed parameters (KV etc) with the ESC_Config_Tool, write, reboot...
- tested communication with the FC (betaflight 4.5.1 dshot600): OK
- checked motor behavior without props (sound, vibration, rotation direction, fc-feedback[speed, temp]): OK
- analog current feedback: OK
- ESC UART telemetry: 4x single Voltage but no total voltage
- ESC UART telemetry: 4x single Current (ca. 12.0A at zero speed), no total value (0.00)
- maiden test with props (hoovering, zero throttle, max throttle): OK
- maiden flight with standard PID and filter settings from the "old" BLHeli32 setup: OK

I could not find anything that "feels wrong" - the AM32.ca configurator did not show the bootloader and software version (no infos inside the four green (matched parameters) ESC-Boxes). Probably only a lookup issue...

Subjective impression compared to the BLHeli32 setup: same responsiveness  but less power (only tested with 15degree timing). So I have tested the auto-timing-advanced feature of the RC3 - and this makes a huge difference compared to the initial values. It is hard for me to compare the dynamic of this version to the "old" BLHeli32 setup but with the advanced-timing option, it was feeling great.

After the second successfully flown battery pack, I have checked the temperatures of the motors (again) and everything was cold.

bhuism, 2024-08-11:
@xagos can you please post a picture of the location of the SWCLK and SWDIO pad location?

xagos, 2024-08-11:
Sorry - I did not took a picture and the ESC is "deep inside the copter" ;) But it is very easy to find next to the uC of each ESC... I did not solder it - i have used test pins with internal springs for the DIO / CK signals

bhuism, 2024-08-13:
No worries, I got it all to work, thanks!

benderl, 2024-08-22:
As I have the E55 still laying on my desk, I can post a picture of the pads. Red is CLK, yellow SWD.
Flashing was quite easy. Thanks to @xagos for the docs! The N-Firmware is already included in the last 2.16-rc3.
[image]

benderl, 2024-08-22:
I still only have the battery connected to the esc. Configuring BF gives me some strange values for motor currents.
[image]
Voltage seems to be fine, but currents should be zero right now. Is this a known issue or will this be ok after connecting the motors?

xagos, 2024-08-22:
I have mentioned exactly the same strange values of telemetry data in one of my feedbacks - I am using the FC-ADC capturing instead...

AlkaMotors, 2024-08-22:
Each esc would need a current shunt to show individual current.  There is no connection between the single current sensor on the board and the individual esc's.  The esc are just reporting a floating unattached pin. Use the current sense direct to the fc from the 'cur' pin. There are iflight blitz e55 g071 single esc's that do have the current sensor attached but the not the 4 in 1.

benderl, 2025-01-26:
I have some strange issues here updating the bootloader to the current version. The board is running fine with the old `STM32_G071_BOOTLOADER_PA6_V8` and all "G071 N" versions of the firmware. But every time I try to install a newer bootloader (v12 or current v13, tried G071 and G071_64k with PA6), the configurator (web and the linux version) are not able to connect. I already flashed with ST Cube Programmer and the unlocker tool, always the same result.

Am I missing something or is there a bug in the current PA6 bootloaders?
From the log of the unlocker I would guess, the G071_64k should be the correct version?
```
Open On-Chip Debugger 0.11.0+dev-00792-g18e7df5f9 (2024-08-05-11:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/tmp/_MEIwXFPli/bootloaders/AM32_G071_BOOTLOADER_PA6_V13.bin
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J45S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.300000
Info : [stm32g0x.cpu] Cortex-M0+ r0p1 processor detected
Info : [stm32g0x.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32g0x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Using bootloader /tmp/_MEIwXFPli/bootloaders/AM32_G071_BOOTLOADER_PA6_V13.bin
FLASH_CR 0xc0000000
unlocking flash
FLASH_CR 0x40000000
unlocking option
FLASH_CR 0x00000000
NEW_OPTR 0xfffffeaa
writing OPTR
disabling protections
send OPTSTRT
option reload
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
reg 0x40022020 0xfffffeaa
reg 0x40022024 0x000000ff
reg 0x40022028 0x80000000
reg 0x4002202c 0x0000003f
reg 0x40022030 0x0000003f
reg 0x40022034 0x000000ff
reg 0x40022038 0x00000000
target halted due to debug-request, current mode: Thread 
xPSR: 0x41000000 pc: 0x0800062c msp: 0x20000fb0
erase 2 bootloader sectors and first fw sector
Info : device idcode = 0x20006460 (STM32G07/G08xx - Rev B : 0x2000)
Info : RDP level 0 (0xAA)
Info : flash size = 64kbytes
Info : flash mode : single-bank
Load bootloader
Info : Padding at 0x08000ff4 with 4 bytes (bank write end alignment)
verify bootloader
Success!
```

xagos, 2025-05-08:
Interesting - today I have updated my E55 Blitz (bootloader v8) from the 9 month old 2.16rc3 to the actual 2.18 --> update on esc 1 2 3 was running successfully - no 4 was corrupted based on missing acknowledge... Now esc no.4 is not reachable at all. So I will try to install a fresh bootloader on all escs this weekend; let's see what happens ;)

xagos, 2025-08-06:
I had to install a new bootloader via STLink...

AlkaMotors, 2025-08-06:
The bootloader is protected.  If ever a flash fails just follow these instructions to recover. You can unplug midway or flash a wrong target and this will get you back. https://youtu.be/XE7uxUA_RqY?si=afwMpsRHhNTAanTd

benderl, 2025-08-06:
@xagos 
Do you remember the version of the bootloader? Is the current release working?