am32-firmware/AM32 #99: Feature request: active break

robustini, 2024-10-07:
As already reported on Discord it would be most useful for use on Plane as in my case to have real active breaking, the current one is almost useless for use on fixed wings.
Obviously it will consume more current but in normal use on fixed wings it is something contemplated, little thick will glide without motor but in those few cases the propeller should stop completely, now obviously with the meager braking action it does not happen.
The active brake should go into action only after the first stop of the motor, not immediately after the ESC is powered up.

robustini, 2024-11-11:
On ESCape32 release 12 it has been implemented and works very well.
I hope it will be done in AM32 as well.

https://github.com/user-attachments/assets/e9aa5c95-5fa7-4b75-8860-c30e8c5b43e2

GiulianoDeMicheli, 2025-10-19:
Is it possible to know if this has been resolved in the latest firmware?