am32-firmware/AM32 #155: Feature Request: add S-bus and F.BUS support for control and telemetry

EmanuelStassar, 2024-12-29:
It would be great if support for S-Bus and F.Bus (FrSky standard for control and telemetry over 1 signal wire) could be added to the code. S-Bus will require a user selectable channel nr (1 to 16) that the ESC should respond to. F.Bus can have up to 24 channels afaik.

r1000ru, 2025-02-03:
Great idea. But need information about protocol for settings number of channel.

EmanuelStassar, 2025-02-04:
I am trying to get that info and will share it with you as soon as I have it.

P.S. I made an error, the second protocol is F.Port, not F.Bus. Sorry for the Mixup. I will try to edit it out.

EmanuelStassar, 2025-02-04:
Here is a start for Sbus: https://github.com/bolderflight/sbus/blob/main/README.md

EmanuelStassar, 2025-02-07:
I contacted my FrSky reseller who contacted FrSky. They are open to help out with implementing F.Port. I don’t want to publish the email address here in the open. Who can I contact to send the info?