am32-firmware/AM32 #157: Does demagnetization compensation exist for the AM32?

AkunEverson, 2025-01-02:
In the process of using the AM32 with a high power motor, the motor will stall when accelerating or decelerating rapidly.
Personal thoughts, non-professional analysis: it is suspected that the motor demagnetization time is too long, which affects the motor commutation, and then leads to stalling or stalling.
Therefore, does AM32 have any solution or optimization measures for this phenomenon?

AlkaMotors, 2025-01-02:
If motor demag time is longer than half a commutation then this means the motor isn't operated within its specifications. Do you know the motor has long demag times? What is the inductance of the motor coils and what sort of voltage are you driving it with?  I have experimented with a few different demag methods and I am still not convinced they should be deployed vs using the proper motor and voltage for the application.

AlkaMotors, 2025-01-02:
Any decent motor will have its specs well published. As long as you stay within the voltage and curelrent limits you should not have to worry about demag . Like this.. https://neumotors.com/brushless-motor-manufacturing/neumotors-4600-series-bldc-motors-500-to-2000-watt-bldc-multicopter-motors/

AkunEverson, 2025-01-03:
> If motor demag time is longer than half a commutation then this means the motor isn't operated within its specifications. Do you know the motor has long demag times? What is the inductance of the motor coils and what sort of voltage are you driving it with? I have experimented with a few different demag methods and I am still not convinced they should be deployed vs using the proper motor and voltage for the application.
> 
> What motor is it you are testing ?

Thank you for your reply!

The length of time to demagnetize the motor, I don't have a specific value to reflect;
Specific parameters of the motor: voltage 12S, 42 poles, 110KV, maximum current 40A.
In addition motor inductance value, I currently do not have the tools, temporarily unable to test.

When I use the AM32 with this motor, when I violently accelerate and decelerate, I occasionally have the problem of motor stalling; later, I paired it with a small motor used by FPV-type drones, and found that there is no such stalling phenomenon.
So, I wonder if the stalling phenomenon only occurs in high-power motors, and therefore consider the demagnetization issue.

Is there a configuration item or optimization for the AM32 that can solve this problem?

AlkaMotors, 2025-01-03:
I hope that there is. I would like to test some new firmware on your setup. Are able to join the discord ? There are a few things that can be done.

AkunEverson, 2025-01-06:
> I hope that there is. I would like to test some new firmware on your setup. Are able to join the discord ? There are a few things that can be done.

Yes, you can. What do I need to do?