am32-firmware/AM32 #172: no telemetry output Vortex F65 ESC

stefanbeier, 2025-02-07:
Hello, I have a problem with a Vortex F65 ESC from OddityRC. It is currently running with AM32. It has AM32_F4A_4IN1_F421_2.17.hex on it. The MCU is an AT32F421-F4AK8U7.
The ESC has a telemetry pin, but no telemetry data comes out.
Do I have to load a different firmware onto it, or does telemetry not work in general with this ESC?

AlkaMotors, 2025-02-07:
Telemetry is not active.  That firmware is not made for the vortex.

stefanbeier, 2025-02-07:
Which firmware fits the Vortex ESC?

AlkaMotors, 2025-02-07:
I really don't know. We don't have a schematic. They never sent us and esc nor did they ever do a pr with their information. They support and promote an abandoned am32 fork on their website.     If they wanted to sort it out they would have to spend the 30 seconds it would take to do so. Unfortunately we just don't know anything about it.

stefanbeier, 2025-03-19:
[image]

Hello AlkaMotors, can you do anything with this information?
Regards, Stefan

AlkaMotors, 2025-03-19:
Telemetry output from the vortex is most likely not pa14. I will have to make a different file for them. There might only be temeperture and rpm available though . Current and voltage sense probably work.

RNewDeal, 2025-03-20:
Unsure how helpful this is but I got their 80A esc working with proper telemetry with this target.

[image]

stefanbeier, 2025-03-20:
Do you have a HEX file for this? I can't find any firmware with that name on Am32.ca.

stefanbeier, 2025-03-22:
It is possible to get the hex file?

RNewDeal, 2025-03-24:
If you join the discord I can send it to you.

stefanbeier, 2025-03-24:
I am to find under stefanbeier_39641 at discord.

AlkaMotors, 2025-03-24:
Can you try the rush firmware? . It's the same pinout except it uses the more standard pb6 telemetry pin.https://github.com/am32-firmware/AM32/releases/download/2.18rc1/AM32_F4A_RUSH_F421_2.18.hex

stefanbeier, 2025-03-25:
The engine is running, but no telemetry data is being output. Not even to PB6.

AlkaMotors, 2025-03-25:
Not even temperature?  Are you requesting telemetry by dshot or do you have the 30ms telemetry option on?

AlkaMotors, 2026-08-25:
Tested and telemetry working as expected. Do NOT use firmware target made for other brands hardware and expect it to work.

closed 2026-08-25 by AlkaMotors

DerGeheimdienst, 2026-08-25:
Hello !

Thanks for the reply !
Which firmware is to be used with the oddityrc Vortex F65 ?
The File AM32_ODDITYRC_F421_2.20.hex does not work for it. It wont arm..

Best regards,
Sebastian

AlkaMotors, 2026-08-25:
Try the F4A_SINGLE_F421 file

AlkaMotors, 2026-08-27:
@DerGeheimdienst That actually narrows it down perfectly.  I think I can guess at the hardware profile. I have also been in contact with oddity so we will get proper targets in there soon and schematics for them.

AlkaMotors, 2026-08-27:
Ok mystery solved.. this is the pinout.. they actually attached pa14 AND pb6 to telemetry. This make none of the output work unless we disable the debug output.  

There is no current input at all.

[image]

DerGeheimdienst, 2026-08-27:
Thank you - 
Maybe they chose the name of the company for a reason.. 
So, there will be no telemetry values for the current, because there is no sensor attached ? 
Is there a sensor on board ?

DerGeheimdienst, 2026-09-11:
I can try out the Firmware if you have no actual ESC on hand.
Best regards,
Sebastian

DerGeheimdienst, 2026-11-09:
Hello ! Has this issue been fixed with the new 2.22 Version ? 
Best regards