am32-firmware/AM32 #173: Issue with flashing CAN Firmware

AhlawatAnup, 2025-02-08:
We have created a custom ESC using the STM32 L431 and have successfully flashed the CAN bootloader onto it. However, we are facing issues when flashing the firmware. We don’t have an ST-Link V3 at the moment, so we are using an ST-Link V2, which doesn’t have a CAN bus adapter.

Is there any way to flash the firmware?

** Some important points to note:

- We have successfully tested Serial Wire (DShot 300), with both the bootloader and firmware successfully flashed onto it with motor spinning.
- - PA2 Signal Pins

thedevanaproject, 2025-08-30:
I know it's an old issue and I'm sure you found a solution in the meantime. For future reference, though, you can flash the firmware either way.
1. The same way you flashed the bootloader, using the ST-Link and STM32CubeProgrammer.
2. Using an Arduino or flight controller connected to the signal pin (PA2 in your case).
3. Ardupilot and Mission Planner apparently have CAN pass-through functionality for flashing.

In any case, don't forget to flash the eeprom_default.bin to address 0x0801F800 for CAN or 0x0800F800 for the 'normal' version.

Variants 1 and 2 definitely work. I haven't managed to successfully test variant 3 yet.

donpurnell, 2026-09-12:
@dmokel wrote "You can directly flash the firmware to an ESC with a CAN-enabled AM32 Bootloader using a DroneCAN Adapter and the DroneCAN GUI Tool." 
But how? I keep running into show-stopper errors :

I have 6 vimdrones S50 CAN ESCs, chained via CAN bus to a vimdrones CAN-to-USB adapter.  The DroneCAN GUI Tool running on Ubuntu 24.04  shows 6 healthy CAN nodes, with the vimdrones CAN bootloader on each.  But attempting to flash the firmware AM32_VIMDRONES_L431_CAN_2.21.bin    to one of these nodes only gives error 
ERROR dronecan_gui_tool.main Node spin error [1 of 1000]: TransferError('Toggle bit value 32 incorrect on frame 2')
MAVLink packet loss 0.00%

I have also tried the DroneCAN Web Tool, but cannot connect. 
Is there a reliable guide anywhere ?
Thanks Don