am32-firmware/AM32 #175: VimDrones DroneCAN ESCs missing from downloads area

rmackay9, 2025-02-18:
On the [ArduPilot wiki](https://ardupilot.org/copter/docs/common-am32-escs.html) we list a few known working AM32 DroneCAN ESCs including [these from VimDrones](https://dev.vimdrones.com/products/vimdrones_esc_nano/) but it seems that the firmware for these ESCs is not on [the AM32 downloads page](https://am32.ca/downloads)

By the way, I'm very happy to add more known working DroneCAN ESCs to that page, just ping me

dmokel, 2025-06-26:
Hi, rmackay~, we publish a dronecan am32 esc, you can find it here: [https://item.taobao.com/item.htm?id=889937578158](https://item.taobao.com/item.htm?id=889937578158)

some picture about this dronecan am32 esc：

[image]

[image]

[image]

Can we list it on the Ardupilot wiki?

dmokel, 2025-07-01:
Hi @rmackay9 , I've create the pr of Pogo_DroneCAN_6S40A_ESC target #224