am32-firmware/AM32 #185: issue with actual_current linear error

475241512, 2025-03-05:
hello,
ESC MCU STM32L432(compatibility with L431), shunt=1mR, current amplifier Gain = 100, and set MILLIVOLT_PER_AMP=100
I plugged the ESC into a resistive load meter ,
the ESC CAN reported current :
0.65A at meter measuring 1A, 
1.43A at meter measuring 2A, 
2.21A at meter measuring 3A, 
2.99A at meter measuring  4A, and so on..
it looks as if there is a linear relationship: 
**measuring Current = (actual_current+0.13)/ 0.78**;
In other words,only if I set the variable **actual_current = (actual_current+0.13)/0.78** before packing the CAN message, I can get the same value as the measured value and the CAN message of the flight control.
when I measure the voltage on the ADC pin, it's about 100mv at 1A and 200mv at 2A, ,,so it seems that the ADC voltage is right.
Is there something wrong with the settings?

AlkaMotors, 2025-03-19:
@475241512 I have not been able to test this. Is the adc accurate for other readings like temperture or voltage?

AlexKlimaj, 2025-04-01:
I think I see the same thing.

battery_status/current_a is measured by an INA238 power monitor. It should be about 0.5A higher than the ESC current.
esc_status/esc.00/esc_current is the total current for the 4 in 1 ESC. 

You can see at lower currents, they match, but once they hit about 28A, the esc reports higher currents.

[image]

https://review.px4.io/plot_app?log=1107c065-5fcb-4180-a826-9bbcbd39d4a3

AlkaMotors, 2025-04-02:
@AlexKlimaj The op has a difference at low current. It looks like the correction factor in your case would make the lower current levels incorrect. The start of your corrected chart looks far off now. It think it would be even be further off at 50a for you. It doesn't look like the same scale of error.

AlkaMotors, 2025-04-02:
Your best bet if help on the discord. We can debug things on there. The first thing is to look at the raw adc values and compare them with measured voltages. If the raw adc values are incorrect then the current sense circuit need to be adjusted. If the target needs a different multiplier for whatever reason than expected the you can just change the target mv/a and offset values.

475241512, 2025-05-20:
> Your best bet if help on the discord. We can debug things on there. The first thing is to look at the raw adc values and compare them with measured voltages. If the raw adc values are incorrect then [...]

It looks like a sampling or calculation problem. Taking the values from the electronic load meter as accurate, I'll redo the test and record the ADC memory values and compare them to the ADC pin measurements and the CAN message values to determine if it's a hardware interface issue, such as low-pass filtering, etc., or if it's a sampling or computation issue.

475241512, 2025-05-20:
> [@475241512](https://github.com/475241512) I have not been able to test this. Is the adc accurate for other readings like temperture or voltage?

Temperature and voltage are extremely accurate. In fact, for the reference voltage, I have tried using both the dynamically measured VDDA value and a fixed 3300 value, with the results being identical.