am32-firmware/AM32 #202: The locally compiled HEX file cannot run properly.

sunyanmeng963, 2025-04-03:
The HEX file compiled by myself contains only one HEX file, while the HEX files in the "release" section contain two HEX files. What are the differences between them? How should I proceed if I want to download the HEX file that I compiled myself?

AlkaMotors, 2025-04-03:
I am not sure what you mean? The releases for am32 have one hex file per esc? Where are you seeing 2?

1weew1, 2025-05-15:
I downloaded the source code for AM32 from GitHub and compiled the firmware myself, but there is no response when I download it to the microcontroller. However, the firmware compiled by the AM32 author works fine. I am using Keil for compilation, and the main control is AT32F421K8U7. Do you know what the reason might be?