am32-firmware/AM32 #209: Telemetry readout via Arduino

ltwin8, 2025-05-10:
Hi there, i would like to read out the AM32 ESC telemetry via an arduino to Display on an OLED
Is there a Library available to do this?

Thank you for any help :)

kaidegit, 2025-05-20:
just solve the packet

https://github.com/am32-firmware/AM32/blob/66960fafca6d4e165c0eaa77fa9a97862ae95965/Src/kiss_telemetry.c#L8-L32

ltwin8, 2025-05-20:
Thanks for the answer, can you give me an example on how to do it?

Best regards

ericlangel, 2025-06-11:
```
while (Serial1.available()) {
      if (count < 10) {
        Kiss[count] = Serial1.read();
        //Serial.print(Kiss[count]);
        //Serial.print(" ");
      }
      else {
        Serial1.read();
        //Serial.print(Serial1.read());
        //Serial.print(" ");
      }
      count++;
    }
    //Serial.println();
    //Serial.print("crc: ");
    //Serial.println(get_crc8(Kiss, 9));
    if (get_crc8(Kiss, 9) == Kiss[9]) {
      rotations = Kiss[7] << 8 | Kiss[8];     // RPM
      rotations = rotations * 100;                  // See Kiss Protocol
    }

uint8_t update_crc8(uint8_t crc, uint8_t crc_seed) {
  uint8_t crc_u, i;
  crc_u = crc;
  crc_u ^= crc_seed;
  for ( i = 0; i < 8; i++) crc_u = ( crc_u & 0x80 ) ? 0x7 ^ ( crc_u << 1 ) : ( crc_u << 1 );
  return (crc_u);
}

uint8_t get_crc8(uint8_t *Buf, uint8_t BufLen) {
  uint8_t crc = 0, i;
  for ( i = 0; i < BufLen; i++) crc = update_crc8(Buf[i], crc);
  return (crc);
}

```

This is an example for getting the rpm

ltwin8, 2025-07-19:
Sorry for just getting back this late...

i had a few hardware issues to sort out...

for quick update i am using an seeed xiao (SAMD21) for this and archived to control the ESC speed via the servo library, i had not found any workaround with dshot...

but the telemetry i still not get...

i would like to read out the RPM, Temp, Voltage and amperage.

I connected the ESC telemetry to the Serial1 RX and checked the setting 30ms telemetry inside the AM32 configurator.

i began the Serial1 on 115200 baud using Serial1.begin(115200); do i need to change something else? like start, stop bit? i have found in Mcu/f421/Src/serial_telemetry.c the baudrate. (usart_init(USART1, 115200, USART_DATA_8BITS, USART_STOP_1_BIT);)

Can you maybe help me out with further help?

Also i would like to program the ESC via the SAMD21 without the need to swap it for an old flight controller just to configure/flash the ESC... is there an example somewhere to do that? i could only find the hacky solution using the external usb-uart converter on 328P type boards...

thank you all for your help

kaidegit, 2025-07-19:
which firmware version is your esc, and which configurator are you using?

there are 3 version of eeprom, and the old configurator and web configurator only support 2.18 and below. if you use master version, you should use [this](https://github.com/am32-firmware/ConfigTool_1.90) new tool.

ltwin8, 2025-07-19:
Got it! 

this variables/defines are necessary:

#define ESC_Serial Serial1
int ESC_V = 0;
int ESC_A = 0;
int ESC_RPM = 0;
int ESC_T = 0;
int ESC_mAh = 0;
int buffer_count = 0;
byte Kiss[20];

And here a sample function that will read in the ESCs telemetry. It does NOT need to be called on every tele packet, it will clear the buffer and take the next ariving package which arrives after clearing.
may most llikely need improvements... but as an example it shoud do

void get_telemetry() {
  ESC_V = 0;
  ESC_A = 0;
  ESC_RPM = 0;
  ESC_T = 0;
  ESC_mAh = 0;
  buffer_count = 0;

  Serial.println("Clearing Buffer... ");
  while (ESC_Serial.available()) {
    Serial.print(ESC_Serial.read());
    Serial.print(" ");
  }
  Serial.println();
  Serial.println("Waiting for ESC... ");
  while (!ESC_Serial.available())
    ;

  Serial.println("ESC Raw Data:");
  while (ESC_Serial.available() && (buffer_count < 20)) {
    if (buffer_count < 10) {
      Kiss[buffer_count] = ESC_Serial.read();
      Serial.print(Kiss[buffer_count]);
      Serial.print(" ");
    } else {
      ESC_Serial.read();
      Serial.print(ESC_Serial.read());
      Serial.print(" ");
    }
    buffer_count++;
    delayMicroseconds(2000); //to compensate for missing USB connection on standalone SAMD21
  }
  Serial.println();
  Serial.print("crc: ");
  Serial.println(get_crc8(Kiss, 9));
  if (get_crc8(Kiss, 9) == Kiss[9]) {

    ESC_T = Kiss[0];                   //Temperature
    ESC_V = Kiss[1] << 8 | Kiss[2];    //Voltage
    ESC_A = Kiss[3] << 8 | Kiss[4];    //Amperage
    ESC_mAh = Kiss[5] << 8 | Kiss[6];  //mAh
    ESC_RPM = Kiss[7] << 8 | Kiss[8];  //RPM
    ESC_RPM = ESC_RPM * 100;           //reported eRPM is 1/100th of the actual eRPM

    Serial.println();
    Serial.println("ESC Telemetry:");
    Serial.print("Temp: ");
    Serial.println(ESC_T);
    Serial.print("Voltage: ");
    Serial.println(ESC_V);
    Serial.print("Amperage: ");
    Serial.println(ESC_A);
    Serial.print("Capacity: ");
    Serial.println(ESC_mAh);
    Serial.println();
  }
}

uint8_t get_crc8(uint8_t *Buf, uint8_t BufLen) {
  uint8_t crc = 0, i;
  for (i = 0; i < BufLen; i++) crc = update_crc8(Buf[i], crc);
  return (crc);
}

uint8_t update_crc8(uint8_t crc, uint8_t crc_seed) {
  uint8_t crc_u, i;
  crc_u = crc;
  crc_u ^= crc_seed;
  for (i = 0; i < 8; i++) crc_u = (crc_u & 0x80) ? 0x7 ^ (crc_u << 1) : (crc_u << 1);
  return (crc_u);
}