am32-firmware/AM32 #215: Arming and Idle Throttle Value Weird Behaviour

Andi7891, 2025-05-27:
Hi,

ESC Model: HAKRC 35A the version with stm32f051 and not F421,etc.
Protocol used DSHOT600
Flight controller firmware: Custom
Same issue: https://github.com/am32-firmware/AM32/issues/125, the author fixed the problem just by treating the 0 as being idle instead of 48 (This I guess removes the option to disarm, unless you cut the dshot signal)

I encounter a weird behaviour with four HAKRC 35A, previously they were running BLHELI32 and everything was fine regarding arming and throttle position, after I flashed them with AM32_DIATONE_F051_2.18 they seem to work fine but the arming happens at 0 throttle and at 48 they start spinning. This behaviour it's really inconvinient for many reasons.

There is an alternative that should be "more compatible" AM32_HAKRC_K_F421_2.18 instead of AM32_DIATONE_F051_2.18, but the problem is that it's for f421 and not f051,maybe with some guidance I could convert the HAKRC fw for f051(Doesn't seem to be many macros needed to change the uC). My assumption is that the precompiled DIATONE fw has a bug (which I can't check, due to no source code available). I might try a previous version of the DIATONE fw.

Does someone else encountered the same problem on the newer fw versions?
Does someone know how for example betaflight deals with am32 ESCs? (They assume 0 as being 48?) I checked Betaflight a bit, but couldn't find anything concrete.

Thanks

frank26080115, 2025-05-27:
I'm the one who filed issue 125. I don't know how betaflight deals with it but what I did was add additional if-statements into ELRS that handles == 0 and != 0 separately, it's not that much work

I would be very surprised if one build of AM32 interpreted the throttle differently from another build just because of a MCU change, that throttle reading code isn't platform-variant.

Andi7891, 2025-05-27:
Thank you very much for your fast answer.

I thought so too that there shouldn,t be any difference between platforms, unless there are more changes than we assume.

 If you could be more precise regarding where to make the change, I would appreciate. 

Thank you.

Andi7891, 2025-05-27:
Regarding LEDs, not even the LEDs work properly they stay all on(so white) :)))
And also because the values under 48 are treated as throttle values, the LEDs can't be turned on and off using commands.

Maybe someone else has an idea where does this problem come from. 

Thank you for the info though, it helps for sure.

AlkaMotors, 2025-05-27:
Just to clear up misconceptions here. 
Dshot 0 is what betaflight sends on power up. If you want motors to stop send dshot 0. This arms both am32 and blheli32 esc's. 
Dshot 0 is also sent before commands by betaflight. 
All dshot values below 48 and above 0 are dshot commands.
Dshot is a digital signal . We don't need two different values for 0 throttle.

AlkaMotors, 2025-05-27:
This also matches how regular servo inputs have always worked. To arm any esc send it 0 throttle. To stop any motor from spinning send it 0 throttle too.