am32-firmware/AM32 #237: Support for HGLRC SPECTER 90A Single ESC

ayandeepd1, 2025-07-27:
Hi,
There is already support for HGLRC esc for f051 but latest  HGLRC SPECTER 90A is based on g071.
Please provide support for HGLRC SPECTER 90A originally on BLheli32_32.9.
(link: [https://www.hglrc.com/products/hglrc-specter-90a-single-esc-high-performance-professional-esc](https://www.hglrc.com/products/hglrc-specter-90a-single-esc-high-performance-professional-esc)

mikerocklewitz, 2025-08-25:
Interesting there is a target for ST_G0_06..........that happens to be the same target name used for blheli32, but what is the "N" designation?...

do all things work? current sensor?

mikerocklewitz, 2025-08-26:
What bootloader did you end up using? I think I'll crack one open tonight and see what happens.

The file with matching name to the blheli32 build seems like it would make the most sense.

ayandeepd1, 2025-08-29:
> Interesting there is a target for ST_G0_06..........that happens to be the same target name used for blheli32, but what is the "N" designation?...
> 
> do all things work? current sensor?

I was playing with the settings a bit for current limit. Setting current limit on this one just makes the motor spin with less energy, the speed or torque seems to do nothing with the current set. I have very little idea on how to load the motor so that the current limit could be stress tested, but seems to me the sensors are not calibrated for am32. 
Also it would be super helpful if you could suggest ways on how to load the motor to stress test the set limits.

AlkaMotors, 2025-08-30:
Chances are close to zero that the current sensor values are going to be correct but we can change them easily. All we need to know is what current the esc says it's putting out when not running (at 0 actual amps) and the current that is reported by the esc at a known current if you have an external current sensor.

HGLRC-T, 2025-09-06:
Hello guys. Any question?

ayandeepd1, 2025-09-06:
> Chances are close to zero that the current sensor values are going to be correct but we can change them easily. All we need to know is what current the esc says it's putting out when not running (at [...]

Could you guide or point towards as to how to log the current sensor values from the esc.
Also finding those 2 values what are the next steps to calibrate?
Thank you

mikerocklewitz, 2025-09-06:
@HGLRC-T 

You could help by providing us with the proper parameters ;)

#ifdef  ST_G0_06_N_G071
#define FILE_NAME "ST_G0_06_N_G071"
#define FIRMWARE_NAME "G0_06_N_G071"
#define DEAD_TIME 50
#define MILLIVOLT_PER_AMP 15
#define CURRENT_OFFSET 0
#define TARGET_VOLTAGE_DIVIDER 110
#define HARDWARE_GROUP_G0_G
#define USE_SERIAL_TELEMETRY
#define SIXTY_FOUR_KB_MEMORY

mikerocklewitz, 2025-09-06:
> Could you guide or point towards as to how to log the current sensor values from the esc. Also finding those 2 values what are the next steps to calibrate? Thank you

There should be a way to read out the telemetry from betaflight cli or similar......we need what its reporting at no load and what it is reporting at a known load (on an amp meter) then im not sure the exact process to get the offset - but thats minor after we have the info.

thegello, 2025-09-07:
I want to make hk3b203 v2.1 (hakrc branded) target, is there any docs how to investigate and configure target?

mikerocklewitz, 2025-09-08:
> I want to make hk3b203 v2.1 (hakrc branded) target, is there any docs how to investigate and configure target?

Definitely not in this issue thread.

mikerocklewitz, 2025-09-13:
Something is weird with this. 

Using the st_g0_06 2.18 build...only sometimes can I get it to connect to the flight controller.  

The esc is manually set to dshot in configurator.  Sometimes I can get esc-configurator/am32 configurator to connect through the flight controller using passthrough, but it is not consistent.  I need to disconnect wires and reconnect, then sometimes it will detect the esc.....and ive never been able to connect to it with bidirectional dshot enabled in betaflight.

Ive attempted serial telemetry in betaflight - i cannnot get any info out of it.  Same with bidi dshot - nothing in cli.  I guess I could hook up the sillyscope and see if anything is truly there, but I've lost momentum.

something is weird with this esc or the am32 build.

mikerocklewitz, 2025-09-14:
> Sounds like something is broken. If an esc is not 100 percent we won't support it.

i agree with that. I'll mess with it some more eventually but i needed a break. 

Fried one of my no-name test flight controllers in the process of trying to figure out if I had a wiring issue or it just wasnt detecting it. didnt kill my computer - learned to use a usb isolator a long time ago 🤣 was not a wiring issue.

ayandeepd1, 2025-09-15:
> It's not worth the time. If I find a bad esc I toss it and move on. There is obviously some electrical fault here.

i have been using this for 2days now to run a 35 series motor under moderate load using 1-2ms pwm signal. Seems to work fine. I dont have a fc to try telemetry but otherwise there doesn't seem to have any operating fault.
I am using bootloader: AM32_G071_BOOTLOADER_PA6_64K_V13
firmware: AM32_IFLIGHT_BLITZ_N_G071_2.19

ayandeepd1, 2025-09-15:
> > i have been using this for 2days now to run a 35 series motor under moderate load using 1-2ms pwm signal. Seems to work fine. I dont have a fc to try telemetry but otherwise there doesn't seem to have any operating fault.
> 
> Digital modes are more important to me than pwm.
> 
> I only tried the file that matches the blheli32 name.
> 
> [@HGLRC-T](https://github.com/HGLRC-T) provided a schematic. If someone writes a target based on what they provided, gives it some random current data so it outputs something, and sticks it in this issue thread (compiled or not) ill test and come up with a dialed in calibration if it works.
> 
> ** If the "st g0 06 n" build i used is proper to the schematic and im having these issues, id say we are screwed. i did not look into this**
> 
> Im not concerned enough to do a ton of work on this esc. It would be nice. But it does work as is.
> 
> The esc i did this with i dont see as defective hardware or broken. I still have debug wires hooked up and I'll play if someone presents something worth playing with.

sounds fair. 
thanks for the support though

AlkaMotors, 2025-09-15:
Blitz_N and g0_06 are identical targets.

SSpicha, 2025-09-17:
> > <img alt="Image" width="1905" height="722" src="https://private-user-images.githubusercontent.com/33262981/490481535-db6479db-b9a3-4924-9a09-55eef2f9f808.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTgxMDM4NTAsIm5iZiI6MTc1ODEwMzU1MCwicGF0aCI6Ii8zMzI2Mjk4MS80OTA0ODE1MzUtZGI2NDc5ZGItYjlhMy00OTI0LTlhMDktNTVlZWYyZjlmODA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTE3VDEwMDU1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4ODQ5YTQxMTFlZGYyNTIyNzBjYzY4ZWE0ZDgwODM2NzQ5MjU3YTYzZWNiN2U4ZWVkZWUxMmU1NGUxOTdkNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JRQvoL7H6ZO5tgQ9spJd1EeW8nfqCLfRFOJz7WlAdRA">
> > Post-Reboot Memory Instability It appears there's an issue with memory or configuration persistence following the first reboot. I observed that after the system is restarted for the first time, several values become incorrect or corrupted. This instability necessitates a complete reset to default settings in order to restore normal operation.
> 
> I think i have also faced this issue. had to send multiple default values to esc to restore flash operation. Did you face the issue again after default value was sent?

Yes, I tried several times, and once with a firmware update. The result is the same. If I select the Auto timing advance option, the engine stops running altogether.

AlkaMotors, 2025-09-17:
Eeprom does not get corrupted. Flashing defaults sets version to 1.35. Upon succesful boot firmware will set eeprom version.

mikerocklewitz, 2025-09-19:
Alright folks.  I have good news.

st go 06 n works fine. culmination of user error led to this.

Current at zero is reading 39.11 amps.   At 0.64 amps of draw i get 40.88