am32-firmware/AM32 #240: SpeedyBee BLHeli_32 50A 4-in-1 ESC Support (stuck on 32.9)

Dumpflam, 2025-08-10:
I know you're propably overwhelmed by esc support requests but most of these speedybee escs are stuck on 32.9 which causes prolems (at least for me and my friends) with re-arming after a crash. I would be very very happy if maybe you guys either modified an existing version to work or made a whole new one.

Dumpflam, 2025-08-17:
> I have flashed my SpeedyBee BLHeli_32 50A 4-in-1 using method from this video: https://www.youtube.com/watch?v=klyASjIe7EI. Works flawlessly.

Thanks mate.

Dumpflam, 2025-09-09:
> DShot telemetry? Yes, absolutely. RPM Filters, dynamic idle, all of it works. I'm not using serial telemetry so I have no idea if that works too.
> 
Does your ESC start synchronized? 2 motors start before the other two for me. The other two sometimes don't even start. I didn't have this problem with blheli32

Dom4n, 2025-09-09:
I had such problem at first, one of them would start out of sync. 
I have reprogrammed once again ESCs and soldered it directly instead of using plugs and now it works reliably.
No problems showed up in ~20 hours of flight time.

4S, SpeedyBee F7 V3 stack. Flashed using JLink EDU because I have only that for ARM based devices - should make no difference. Firmware version 2.18, target AM32_SPEEDYBEE_65A_F421